CrealityOfficial/Ender-3S1 #2: Marlin source code for Sermoon V1

centertrim, 2022-03-08:
Hello,

Can you please post the source code for the Marlin firmware of the Sermoon V1?

Thank you!

Creality-mandy, 2022-03-10:
Hello, so it means you want to DIY Sermoon V1? right? We are designed fully enclosed machine for education , but not for DIY, so we don't post the source code at this time for Sermoon V1. But there is possibility to open it in the future.

centertrim, 2022-03-19:
Yes I am using it for DIY, as are many others.  Enclosed printers are useful when there are cats or other animals that could interfere with the printer.

Please post the Marlin firmware source code as required by the GPL license.  I would like the ability to tweak the firmware for my own needs.

closed 2022-04-22 by Creality-mandy

duchengyao, 2022-05-22:
Are there any plans to update the firmware to support the linear advance version?
@Creality-mandy

duchengyao, 2022-10-09:
Official source code has been published: https://github.com/CrealityOfficial/Sermoon-V1

ffnull, 2022-10-09:
> Official source code has been published: https://github.com/CrealityOfficial/Sermoon-V1

Thnx. I know that becouse its me write directly to creality and ask to provide sources. Also im waiting sources for DWIN lcd.