CrealityOfficial/Ender-3S1 #4: Source code for Ender 3 S1 Pro

attila85, 2022-03-19:
Hi!

Is the source code for S1 PRO available as for S1?

Thanks!

Creality-mandy, 2022-04-13:
Yes, the Pro version is available for S1, but if you only change software, without upgrading hardware, the temperature difference will become large. So we suggest you to upgrade into all mental throat tube.

closed 2022-04-13 by Creality-mandy

attila85, 2022-04-13:
Hi, I do have an S1 Pro, so the hardware is already upgraded. Where can I find the source code?

Creality-mandy, 2022-04-15:
Hi, you can use S1  source code, it has the same parameters. 
https://github.com/CrealityOfficial/Ender-3S1/blob/main/Marlin/Configuration_adv.h

attila85, 2022-04-21:
> Ender 3 S1 Pro touch display is not working if you build firmware from S1
> source code.
> What should be changed to make it working ?

Same issue for me.

Creality-mandy, 2022-04-22:
We are not opening S1Pro screen source code at this moment. when use s1 version to upgrade. Not only to upgrade hardware, but also need to upgrade the software. If possible, we will send the screen produced file to you private by email. How about it?

spectrality-github, 2022-04-22:
I also have stock Ender 3 S1 Pro and I'm not upgrading anything for now.

I only need S1 PRO firmware with Host Actions enabled for Octoprint
Uncommitted line in Configuration_adv.h
`//#define HOST_ACTION_COMMANDS`
Maybe you can release official S1 PRO firmware with Host Actions enabled ?

But it would be nice to have full source code for S1 PRO to do other modification by ourself.

So if I modify configuration.h from S1 source code to much S1 PRO temperatures
Change line in configuration.h
`#define HEATER_0_MAXTEMP 275`
to
`#define HEATER_0_MAXTEMP 315`
enable Host Actions in configuration_adv.h and build firmware from that, you will be able to send me privately matching S1 PRO Touch screen firmware and this will work?

How  I can share my private email to you(@Creality-mandy) privately ?

theSt33v, 2022-05-24:
> Hi, you can use S1 source code, it has the same parameters. https://github.com/CrealityOfficial/Ender-3S1/blob/main/Marlin/Configuration_adv.h

This cannot be true. The S1 source code uses thermistor preset 1 (see Marlin/src/module/thermistor/thermistor_1.h). The highest value in the temptable for this thermistor is 300 C. Since the S1 Pro has a maximum user-selectable temperature of 300 C, the code must be different for the S1 Pro because the maximum temperature in the temptable needs to be higher than the maximum user-selectable temperature. Which thermistor temperature preset is used for the S1 Pro? Or is it using custom thermistor settings? If so, please provide these settings.

jklap, 2022-06-23:
> We are not opening S1Pro screen source code at this moment. when use s1 version to upgrade. Not only to upgrade hardware, but also need to upgrade the software. If possible, we will send the screen produced file to you private by email. How about it?

I would appreciate this @Creality-mandy @CrealityTech !

Creality-mandy, 2022-06-27:
> I also have stock Ender 3 S1 Pro and I'm not upgrading anything for now.
>
> I only need S1 PRO firmware with Host Actions enabled for Octoprint Uncommitted line in Configuration_adv.h `//#define [...]

My email address is zhujinyun@creality.com

mriscoc, 2022-07-21:
The touchscreen of the S1 Pro is not supported.

Pethical, 2023-01-01:
>  We are not opening S1Pro screen source code at this moment. when use s1 version to upgrade. Not only to upgrade hardware, but also need to upgrade the software. If possible, we will send the screen produced file to you private by email. How about it?

**No one asked you to open the screen source code**, just publish **the mainboard firmaware** what is **based on Marlin** and without you provide the source, **you violates the GPL**. Yes, you must, and really **must provide the source with your modifications**. What do you think about you are violating GPL? **Is this ok for your company?**

trapper-, 2023-01-05:
@Creality-mandy A PR was merged to enable HOST_ACTION_COMMANDS for the S1 https://github.com/CrealityOfficial/Ender-3S1/issues/12

But no new firmware release for the S1 Pro yet. Is this coming soon?

We are unable to build it ourselves without access to the source code...