CrealityOfficial/Ender-3S1 #6: Source for STM32F401 board

theSt33v, 2022-04-06:
Please upload the missing source code files for the STM32F401 board (pins_CREALITY_S1_F401RC.h, for example)

Creality-mandy, 2022-04-13:
Hi thanks for your question, please check at the main.

theSt33v, 2022-04-13:
Thank you, but there is still missing code. Please provide an entry for the STM32F401RC_creality build environment in /ini/stm32f4.ini

Creality-mandy, 2022-04-15:
https://github.com/CrealityOfficial/Ender-3S1/pull/7/commits/b94856bbb5ff06bab413abca00535f46dea4a0df

theSt33v, 2022-04-15:
> [b94856b](https://github.com/CrealityOfficial/Ender-3S1/commit/b94856bbb5ff06bab413abca00535f46dea4a0df)

Yes, I see that you have added pins_CREALITY_S1_F401RC.h. However, I still cannot build this firmware because the required build environment, STM32F401RC_creality, is missing from the file located in /ini/stm32f4.ini. Please update /ini/stm32f4.ini so that it contains the STM32F401RC_creality build environment.

theSt33v, 2022-04-15:
Thank you! With the most recent update, the source code is now complete.

closed 2022-04-15 by theSt33v