CrealityOfficial/Ender-3S1 #8: Cannot Flash STM32F4 Ender 3 S1 Boards

houseofbugs, 2022-04-20:
I just received a bunch of Ender 3 S1 boards with the F4 CPUs and NONE of them will take firmware at all. I've tried the code here as well as 3rd party code, none of them will flash.

Please let us know what the proper method to flash these boards with the F4 CPU should be because none of the ones I have will take any firmware updates over SD.

houseofbugs, 2022-04-20:
[
[image]
](url)

fabiomaticus, 2022-04-20:
I have same problem with two Ender 3 S1 with chip F4.
How i do firmware upgrade?

theSt33v, 2022-04-20:
Try creating a folder on your SD card called STM32F4_UPDATE. Place the firmware.bin file in that folder, then perform the normal flash procedure (power off printer, insert card, power on printer).

houseofbugs, 2022-04-21:
Yeah it needs to be in the folder STM32F4_UPDATE and named different EVERY time. Firmware.bin will not work as that is what they used at the factory and its stored in the CPU memory. I have working firmware posted here: https://www.th3dstudio.com/hc/downloads/unified-2-firmware/creality/creality-boards-creality/creality-ender-3-s1-firmware-v2415_301-board/

closed 2022-04-22 by Creality-mandy