CrealityOfficial/Ender-3S1 #9: ender 3 s1 F4 mainboard stuck

Nbrooks639, 2022-05-27:
i am now on my second f4 mainboard that is stuck with the last firmware flashed to it. i have updated it, with correct firmware along the way, now i cant change anything it wont flash anymore

mriscoc, 2022-05-27:
That very strange, then the bootloader is being overwritten, it is very bad because the bootloader must to take care about that, it is out of control of the firmware.

CrealityTech, 2022-06-28:
> i am now on my second f4 mainboard that is stuck with the last firmware flashed to it. i have updated it, with correct firmware along the way, now i cant change anything it wont flash anymore

can change the start address,modified to 0x80000,and use usb prot download with stm flash download
or else give us mail address we send it

Nbrooks639, 2022-07-01:
> 

ok can i send you my address through another link of some sort please?

Nbrooks639, 2022-07-19:
> can change the start address,modified to 0x80000,and use usb prot download with stm flash download or else give us mail address we send it

brooks639@yahoo.com

CrealityTech, 2022-08-17:
> 

hi ,we have send file to you mail,

Nbrooks639, 2022-08-17:
> i have not received anything @CrealityTech

Nbrooks639, 2022-08-19:
Still have not seen any support for this situation. Going on how long now? I am on stm32f4 mainboard number 3 because I can't get a response fast enough for aupport

ffnull, 2022-09-16:
> Still have not seen any support for this situation. Going on how long now? I am on stm32f4 mainboard number 3 because I can't get a response fast enough for aupport

"Great" support

Nbrooks639, 2022-09-24:
> > Still have not seen any support for this situation. Going on how long now? I am on stm32f4 mainboard number 3 because I can't get a response fast enough for aupport
> 
> "Great" support

yea its crazy. i gave up. what keeps happening to yours? samething? after so many flashes it just stops taking firmware?

CrealityTech, 2022-09-27:
> dified to 0x80000,an

sorry about that ,you have not received ? ok i send you email again,please check you email,3QU

Nbrooks639, 2022-09-29:
> > dified to 0x80000,an
> 
> sorry about that ,you have not received ? ok i send you email again,please check you email,3QU

your email makes absolutely no sense what so ever

CrealityTech, 2022-10-14:
> > > dified to 0x80000,an
> > 
> > 
> > sorry about that ,you have not received ? ok i send you email again,please check you email,3QU
> 
> your email makes absolutely no sense what so ever

common don't say that,we have sent the file that you want twice,haven't received yet? 
the email address maybe not crealithTech,please check you inbox with xxxx@creality.com

Nbrooks639, 2022-10-15:
the mainboard does not do anything. ive tried your info and it will not update at all

CrealityTech, 2022-10-21:
> the mainboard does not do anything. ive tried your info and it will not update at all

the update file must in the folder named"STM32F4_UPDATE",

Nbrooks639, 2022-10-21:
yes i know this i have already explained that i did a few firmware updates, i know exactly how it works.

CrealityTech, 2022-10-24:
> yes i know this i have already explained that i did a few firmware updates, i know exactly how it works.

and now how it's go?

Nbrooks639, 2022-10-24:
no it does nothing the board will not change at all its completely stuck like i have been explaining for the passed 6 months this is crazy look how long its been. please send me a new mainboard

FulminatingGhost, 2022-12-04:
Hello,
I am currently facing the same issue. I can not flash new firmware via the bootloader no matter what I try. When I monitor serial data on startup, I sometimes see a message like "Bootloader SD card ..." 
I will try to investigate and re-flash with a ST-Link next week.

FulminatingGhost, 2022-12-09:
Well, I was able to flash the firmware and bootloader using an ST-Link. Updating the firmware through the bootloader worked once after that. It would not update on subsequent tries. I also took a flash dump before re-flashing and except for the last filename at offset 0x0800c000 nothing was different from before the re-flash.
Does the bootloader check any other addresses in flash that might stop it from doing the update?

mriscoc, 2022-12-09:
> Well, I was able to flash the firmware and bootloader using an ST-Link. Updating the firmware through the bootloader worked once after that. It would not update on subsequent tries. I also took a [...]

Maybe it could be a bug in the firmware filename checking? Other boards do not allow flashing firmware with the same name as the currently installed one.

FulminatingGhost, 2022-12-10:
> > Well, I was able to flash the firmware and bootloader using an ST-Link. Updating the firmware through the bootloader worked once after that. It would not update on subsequent tries. I also took a flash dump before re-flashing and except for the last filename at offset 0x0800c000 nothing was different from before the re-flash. Does the bootloader check any other addresses in flash that might stop it from doing the update?
> 
> Maybe it could be a bug in the firmware filename checking? Other boards do not allow flashing firmware with the same name as the currently installed one. Is the filename null-terminated?

I always verified the filename to be different from the one before. I also tried using very short or very long filenames to no success.
In flash, there is no null-termination, but rather the remainder of the 256 byte area is filled with 0xFF bytes.

nadavhury, 2022-12-17:
> Well, I was able to flash the firmware and bootloader using an ST-Link. Updating the firmware through the bootloader worked once after that. It would not update on subsequent tries. I also took a [...]

Hi, Can you please explain how you've managed to flash the new firmware with st-link? I'm trying to flash my ender3 s1 pro from the 2.0.8.15F4 version to the new version 2.0.8.24F4 with the laser with no success.
I have tried any sd card I have, (right now the only 8GB card I have is the one supplied with the printer), in the STM32F4_UPDATE folder and outside of it, 4092 allocation units, default allocation size, absolutely tried anything I could find on Reddit, and here, not flashing any firmware, frustrating.
Any help is much appreciated!

FulminatingGhost, 2022-12-19:
> Hi, Can you please explain how you've managed to flash the new firmware with st-link? I'm trying to flash my ender3 s1 pro from the 2.0.8.15F4 version to the new version 2.0.8.24F4 with the laser [...]

Hi,
I used the st-link command line utility to first flash the bootloader at 0x08000000 and afterwards the firmware at 0x08010000. The connector for the st-link adapter is J201 on the CR-FDM-v24S1_301 board. The pinout is, beginning at the side where the SD card and usb connector reside: VCC, DIO, CLK, GND. The VCC must be connected to the VAPP pin of the ST-Link.
If the FW does not start, then the bootloader might want to do an update via sd card.
In the end I resorted to not using the bootloader anymore since the SD update works very unreliable for me.

nadavhury, 2022-12-19:
> Hi, I used the st-link command line utility to first flash the bootloader at 0x08000000 and afterwards the firmware at 0x08010000. The connector for the st-link adapter is J201 on the [...]

Thank you!
The instructions you have written are some really pro stuff,
I have some homework to do :)
Thank you again.

sidcha, 2023-08-01:
Okay, I managed to figure out the issue. I have an ender 3 S1 Pro so everything below applies to that board exclusively. Here are my findings:

- VCC not being connected is not an issue.
- There are 2 unsoldered pins right next to the annoyingly bright blue LED. One of them is RST and the other is GND. You can short these to reset the device (see below).
- st-flash fails due to two reasons:
   - stm32f401 has 256k flash and that is not what it is expecting. You can pass --flash 256k to st-flash to overcome this issue.
   - It also wanted the device to be held in reset. This can be done by shorting the unsoldered 2 pins mentioned above.
- openocd does not have any of the above issues that st-flash has :) So you can just do:
     ```
     openocd -f interface/stlink.cfg -f target/stm32f4x.cfg -c "program klipper.bin verify exit 0x08000000"
     ```
     Note: this removed the bootloader (I don't care about it).
- The 10 pin box connector has a UART line, pins: 3: TX 4: RX 9: GND.
- Right next to the exposed USB-C connector there is are unsoldered 4 pins where you can solder a USB-A connector and get the onboard USB-Serial connection (handy if you don't want to use the USB-C connector).

This is a markup I created for my reference:
[image]

CubeCmder, 2024-01-26:
> Okay, I managed to figure out the issue. I have an ender 3 S1 Pro so everything below applies to that board exclusively. Here are my findings:
> 
> * VCC not being connected is not an issue.
> * There are 2 unsoldered pins right next to the annoyingly bright blue LED. One of them is RST and the other is GND. You can short these to reset the device (see below).
> * st-flash fails due to two reasons:
>   
>   * stm32f401 has 256k flash and that is not what it is expecting. You can pass --flash 256k to st-flash to overcome this issue.
>   * It also wanted the device to be held in reset. This can be done by shorting the unsoldered 2 pins mentioned above.
> * openocd does not have any of the above issues that st-flash has :) So you can just do:
>   ```
>   openocd -f interface/stlink.cfg -f target/stm32f4x.cfg -c "program klipper.bin verify exit 0x08000000"
>   ```
>   
>   
>       
>         
>       
>   
>         
>       
>   
>       
>     
>   Note: this removed the bootloader (I don't care about it).
> * The 10 pin box connector has a UART line, pins: 3: TX 4: RX 9: GND.
> * Right next to the exposed USB-C connector there is are unsoldered 4 pins where you can solder a USB-A connector and get the onboard USB-Serial connection (handy if you don't want to use the USB-C connector).
> 
> This is a markup I created for my reference: ![pinout](https://private-user-images.githubusercontent.com/2995609/257636598-6085a12d-d77e-4243-9aeb-e84a27fc1a51.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDYzMDMzODQsIm5iZiI6MTcwNjMwMzA4NCwicGF0aCI6Ii8yOTk1NjA5LzI1NzYzNjU5OC02MDg1YTEyZC1kNzdlLTQyNDMtOWFlYi1lODRhMjdmYzFhNTEucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDEyNiUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDAxMjZUMjEwNDQ0WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9ODg2YTA3MGNiNjMzNWIyZTY0OWIwYWQ1OGM0MzMyNWIyZDZkZTMyNGQ3NGJlMWI1NDE1ODE0Zjg2NDk2ODVjMSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.ktV3tTPtpgDfKppAjjF0PfrXCrewmcD3S4dcy-3FUFQ)

I am having a similar issue where i cant flash any new firmware to the board. Were you able to flash a new firmware this way? So basically you bridge the RST pin with ground to put it in reset and connect the SWD pins to the dongle? Did you externally power the board while performing the update?

sidcha, 2024-01-26:
> Were you able to flash a new firmware this way?

Yes.

> So basically you bridge the RST pin with ground to put it in reset and connect the SWD pins to the dongle?

Yes.

> Did you externally power the board while performing the update?

No, you don't have to.

mriscoc, 2024-01-27:
For future reference, we identify some kind of flag zone in the EEPROM for the Ender3S1 F4 bootloader, if that zone is written with certain values the bootloader refuses to flash new firmware. That zone seems to be between the EEPROM index 916 and 926, so custom firmware must avoid that flag zone in the EEPROM when load and save parameters.

https://github.com/mriscoc/Ender3V2S1/issues/1180#issuecomment-1911528544

CubeCmder, 2024-01-27:
I haven't been able to store any settings to eeprom, could this explain that? Also am supposed to run the gcode and then perform the normal flashing sequence?

mriscoc, 2024-01-27:
> I haven't been able to store any settings to eeprom, could this explain that? Also am supposed to run the gcode and then perform the normal flashing sequence?

It depends on which firmware your printer currently has but you can try it. Put the firmware in the STM32F4_UPDATE SD card folder, run the G-code file and reboot the printer.

CubeCmder, 2024-01-27:
I just applied this fix and it worked! Thanks again Miguel!

dkostwg, 2024-06-06:
Hey, I know this is an old thread but hopefully someone is still is available for a question. I have an ender 3 S1 with an F401 RCT6 chip. I can use STlink V2 USB dongle to program the STM32f401RCTx_BootLoader_v1.bin bootloader @ 0x08000000 and then program the Klippy.bin generated with make flash @ 0x08010000. The programming says it completes fine using STM32cube programming SW. However when connecting to my pi 3B (running latest klipper/mainsail/etc., the printer just not recognized by Klipper and it sits at trying to connect. lsusb does not show anything besides a generic USB interface so at least it sees that tied to the board when plugging in USB  but it does not know its the printer and it does not appear a DFU device. So I am guessing perhaps there are other parameters to write to the FLASH(?) to let the bootloader know some additional information. The STM32F401 FLASH was fully erased at some point while I was experimenting with trying to get it working. Any ideas? Thank you!

mikechristiansen, 2024-11-23:
I have this same issue, corrupt bootloader. I have an STLink but cannot find the STM32f401RCTx_BootLoader_v1.bin file. Can anyone point me to where to get this? @CrealityTech ? Thanks!

GitNuked, 2025-10-31:
I don't mean to necropost even further but it would seem I'm at a bit of a dead end with @CarCar-T  's solution to interfacing with the mainboard as everytime I do I'm hit with data read fail - I've also tried shorting the reset pin with the ground provided. I've been able to read the board in this state but not write. If anyone could provide further insight to a noob like me I'd be over the moon, thanks y'all!