CrealityOfficial/Ender-3S1 #12: Ender-3 S1 Pro Firmware HOST_ACTION_COMMANDS

wassereimer86, 2022-08-20:
I am using Ender-3S1_Pro_HWv24S1_301_SWV2.0.8.24F4_F401_FDM_LASER from 3 Aug. 2022. HOST_ACTION_COMMANDS are not enabled. Will you enable it again? It was working with older releases, a few people on the internet say.

Sadly there is no repository for the Ender-3 S1 Pro and i can´t find the current configs for the printer to compile a firmware by myself. Can you provide them if you not want to enable HOST_ACTION_COMMANDS?

mdlopresti, 2022-09-20:
I would also like this, PR attached that should enable it the next time a firmware is built

CrealityTech, 2022-09-27:
this firmware has merged

closed 2022-09-27 by CrealityTech

motiregev, 2022-12-03:
@CrealityTech  I download the last version from https://www.creality.com/pages/download-ender-3-s1-pro  which is Ender-3 S1_Pro_HWv24S1_301_SWV2.0.8.24F4_FDM_LASER  as it say in the site but the actual version in the bin file is: 3s1pro_hw24s1_301_V2.0.8.16F1_F103_LASER_FDM.bin (it was relaesed in 29-11-2022).
1. It seems to be mismatch in the version between the site and the the inner bin file).
2. In this version the HOST_ACTION_COMMANDS are stil disabled.
When will we have a version that include it for Ender-3 S1 Pro?

trapper-, 2022-12-25:
I'm also waiting for this release, thanks!