CrealityOfficial/Ender-3S1 #16: Ender3 S1 Plus

g2b1980, 2022-10-05:
Is it possible to get the firmware for the Ender3 S1 Plus along with the modification of enabling HOST_ACTION_COMMANDS ?

wizardkitty, 2022-10-25:
Well they have added HOST_ACTION_COMMANDS to the firmware but from what I know the touchscreen is closed source and that's why they don't put firmware out for the pro and plus. But you could compile this with some edits but you would most likely loose touchscreen functionality from what I've heard.

wizardkitty, 2023-01-15:
> Is it possible to get the firmware for the Ender3 S1 Plus along with the modification of enabling HOST_ACTION_COMMANDS ?

It now can be done. They released the source. If you give me an email i would be willing to send the bin files for whatever version chip you have.

joepadmiraal, 2023-02-06:
Hi @wizardkitty I'd love to try out the firmware for the S1 Plus with the HOST_ACTION_COMMANDS.
I have the STM32F401 chip in mine.

wizardkitty, 2023-02-07:
> Hi @wizardkitty I'd love to try out the firmware for the S1 Plus with the HOST_ACTION_COMMANDS. I have the STM32F401 chip in mine.
[firmware-20230207-114150.zip](https://github.com/CrealityOfficial/Ender-3S1/files/10679255/firmware-20230207-114150.zip)
Best you download a creality firmware then replace the bin file in the update folder with this bin file and it should work. It should be built for the s1 plus. if issues arrise just revert to creality stock firmware.

joepadmiraal, 2023-02-07:
@wizardkitty thanks!
In the meantime I was finally able to compile it myself with Pethical's branch plus a couple of [small casing related fixes](https://github.com/Pethical/Ender-3S1/pull/5).
I enabled these lines:

``` 
#define ENDER_3S1_PLUS
#define HOST_ACTION_COMMANDS
#define HOST_PROMPT_SUPPORT
```

Anything else I should enable?

When I press the stop button on the printer, Octoprint does not seem to understand that and will continue printing mid air.
Do you by any chance know how to fix that?

Other than that Octoprint seems to work fine, including bedleveling visualization :)

jmachuca77, 2023-02-08:
@joepadmiraal does the display work? I was able to compile with no issues but the display is stuck on the startup screen. I think its a version mismatch. What firmware did your printer have originally? (mine came with 2.0.8.25, and the source code here is for 2.0.8.24 so I think that's why my display does not work)

joepadmiraal, 2023-02-08:
Hi @jmachuca77.
The display works fine.
I had firmware version 2.0.8.23F4 plus screen v1.0.1 installed when I installed my own version.
I did the update by placing the `firmware-......bin` file in the  `STM32F4_UPDATE` directory on the sd card.
After turning the printer on, the display did freeze for a while with some dots of the logo visible.
After less than a minute the normal menu appeared.

What kind of printer and cpu do you have?

jmachuca77, 2023-02-08:
S1 Plus, I think they all have the same CPU i just got it yesterday. The firmware built wit no issues once i changed the config file and it seems to be loading the firmware but the display just freezes with the dots. I was able to flash the .24 firmare back and now its working again.

jmachuca77, 2023-02-09:
@joepadmiraal I got it working, I missed changing the define for my processor. Once I changed that now I have it working.