CrealityOfficial/Ender-3S1 #22: [Firmware Request] Unified Bed Leveling

lin-ycv, 2022-11-02:
I own multiple Ender 3 S1 Pro, I've noticed that on some machines ABL works fine, others not so great, any chance the bed leveling menu can include UBL ability and view mesh ability? something like jyers firmware?
[image]

ayushsharma82, 2022-11-09:
I would have added it myself in the screen if the source code was made available.

wizardkitty, 2022-11-15:
UBL would be good to have overall but due to creality making some stuff closed source I cannot make a working firmware version that would be able to work, but if creality open sourced some stuff so where we can edit features of the ui we would be good to go.

jplmmwolters, 2022-11-16:
Yes, would like to have a open source firmware for de LCD screen. @CrealityTech @Creality-mandy

wizardkitty, 2023-01-15:
This could now be possible. Source Code released for the s1 pro and plus. 

Base implmentation would most likely break some stuff with the touch screen for abl but if possible a upgrade could occur for the S1 Pro and plus with ABL and so on.