CrealityOfficial/Ender-3S1 #23: [Feature Request] Enable Linear Advance

ayushsharma82, 2022-11-06:
Ender 3 S1 Pro is very capable machine but the rounded corners at higher speeds is an problem which can be solved by enabling Linear Advance in Marlin.

I would like Creality to add this “Linear Advance” feature and create a release for everyone. At the price point which this printer is getting sold, I would expect it to be enabled from the factory, but this was not the case.

Pic of the issue which occurs at high print speeds without linear advance:
[image]

ayushsharma82, 2022-11-09:
@CrealityTech I know Sonic Pad with Klipper is good but there are a lot of settings which are missing in the firmware and the company is not providing the source code ( which is based on Marlin GPLv3 License ). By law, you guys should have released the firmware on the launch date of “S1 pro” itself as GPLv3 license requires you to disclose the source.

I’m only frustrated with the fact that a machine like Ender 3 S1 Pro has so much capability and it is only held back by its firmware. I don’t mind if you expand the feature set and keep the firmware closed-source but there has been zero useful development so far in the firmware.

Creality should either listen to the community on what they want in the firmware or provide the source code through which people can add features themselves.

Gemerus, 2022-11-14:
@CrealityTech It's absolutely insane that you are offering to buy a device I don't need for 150 bucks to activate one feature that I used on your printer, which cost 250 bucks in total. I don't need a sonic pad, I just need an LE, am I asking for so much? The big mistake was to believe you and buy your product.

jplmmwolters, 2022-11-16:
Yes, would like to have a open source firmware. @CrealityTech @Creality-mandy

kamoteshake, 2022-12-01:
yeah. they have 2 choices, be active with these feature requests for the S1 Pro and Plus or just open source the firmware to the community. I've seen some feature requests here that have been open for a while and no one from @CrealityTech @Creality-mandy responded. Not even a respond on why they won't add things.

I don't know where to voice out that I want the firmware for small tweaks I want to do without changing to a different firmware and making the screen useless.

StrangeAlex3003, 2022-12-06:
> Hight speed printing need creality sonic pad,this well be best combo

Has your sonic pad already learned how to work with your CV-Laser module? I bought a printer and bought a module. I want to use both. But you need a klipper or LA firmware to use the printer properly. And for the laser I need the original firmware. Are you suggesting that I have to reflash the device once a day? Is it so hard to include LA support in the firmware to make all users happy?

Pethical, 2023-01-05:
@CrealityTech 
> Hight speed printing need creality sonic pad,this well be best combo

It is really kind and honorable of your company to realize the mistake and compensate customers with a free sonic pad. Please, also send to me one sonic pad (I purchased an ender 3 s1 pro, what advertised as out of the box high speed print machine). Also please respond my emails and send me the source code of the firmware (is GPL licensed, you should and must to send).

Thank you very much for your support!

Fredlepp, 2023-03-26:
> yeah. they have 2 options here, be active with these feature requests for the S1 Pro and Plus or just open source the firmware to the community. I've seen some feature requests here that have been open for a while and no one from @CrealityTech @Creality-mandy responded. Not even a respond on why they won't add things.
> 
> I don't know where to voice out that I want the firmware for small tweaks I want to do without changing to a different firmware and making the screen useless.

me too, i'm asking to buy a basic ender 3 lcd screen, to let me modifying firmware

Pethical, 2023-03-26:
Hi,
The fw is open source. Creality published the sources

Fredlepp, 2023-03-26:
> Hi, The fw is open source. Creality published the sources

yes but many options you need a menu on the screen. or, creality with this crappy screen, leaves no freedom

DjordjeMandic, 2023-10-04:
@CrealityTech I see that you can enable linear advance in marlin officially but you just don't want to. You enabled input shaping but linear advance is still off.. Most of the benefit would be from linear advance because problems start at 40-50mm/s already.

I'm quoting your latest changelog

-----------------------------------------
Firmware update contents:
   - New model picture preview function
   - **Support high-speed printing** (160mm/s) & optimization of vibration pattern
   - Support automatic PID setting
   - Display/edit leveling data
-----------------------------------------

What do I have from having input shaping and high printing speeds when my corners are going to look like Bibendum.

Yeah I will switch to klipper at some point. I will probably go with DIY variant or BTT Pad 7 because its open and I have control over my hardware without it being overpriced. Remember that Ender 3 series is meant to be **cheap**!!!!!