CrealityOfficial/Ender-3S1 #24: Firmware for S1 Pro?

ayushsharma82, 2022-11-06:
Hello,

Is this firmware used for S1 Pro as well?

Regards,
Ayush

ayushsharma82, 2022-11-08:
@Creality-mandy @CrealityTech , can you guys provide me source code of S1 Pro? I would like to tweak the firmware for personal use ( Marlin only ).

Best would be to open-source it, there’s a whole group on Facebook complaining about this. Which is really a negative reputation for Creality.

dremc, 2022-11-09:
> @Creality-mandy @CrealityTech , can you guys provide me source code of S1 Pro? I would like to tweak the firmware for personal use ( Marlin only ).
> 
> Best would be to open-source it, there’s a whole group on Facebook complaining about this. Which is really a negative reputation for Creality.

The change for the S1 Pro is increase bed/nozzle max temperature and updated PID value for thermistor (PID). Which can be easily update. 

The touch screen on S1 Pro is custom(close source probably) and probably why there not releasing it, but if firmware based on Marlin they should release the source under GPL. 

In configuration.h (CRTL-F and find those line and update values) 

`#define HEATER_0_MAXTEMP 300  `

`#define BED_MAXTEMP      115`

wizardkitty, 2022-11-09:
I would like firmware for s1 pro to open source but I can understand wanting to keep the screen closed but it would be nice for it to be open, that would make alot of people happy and allow for community firmware development without requiring hardware modifications to replace the screen with a know encoder screen.

Gemerus, 2022-11-14:
Wow. I've been using your products for many years, since the release of ender 3, and I've bought it for the sake of OPEN SOURCE. What went wrong guys? I'm glad you made a sonic pad, but I don't need it at all..

wizardkitty, 2022-11-14:
We just want the source code for the ender 3 s1 pro and be able to modify it and not loose function of the touch screen. Open sourcing it would be nice even if it could spur clones.

punchednthenose, 2022-11-14:
We really need the source code!!! You guys call it an open source project but don't give up the coding!! Why???

wizardkitty, 2022-11-14:
@CrealityTech Open sourcing the Touch Screen and S1 pro and S1 plus firmware would allow for more community support, I would love to help enhance the firmware to make it better for everyone but without the source for both I cannot test features with a stock s1 pro nor add features that might require human interaction.

Gemerus, 2022-11-14:
@CrealityTechIt's really quite...weird. I bought a MORE EXPENSIVE version of the printer, and as a PENALTY for that, I can't access its firmware. I could buy the NOT PRO version, and modify the firmware however I want. I can do it on my old Ender 3 PRO, but not on my new Ender 3S1 PRO. Discouraging.

kamoteshake, 2022-11-24:
Just going to comment here to +1 this. I just got an Ender 3S1 Plus.

L1ghtingBolt, 2022-11-28:
> Just going to comment here to +1 this. I just got an Ender 3S1 Plus.

Me too... It's sad I can not change it, because I upgraded it with an all metal hotend

kamoteshake, 2022-11-29:
> Me too... It's sad I can not change it, because I upgraded it with an all metal hotend

huh. I was actually planning to upgrade my hotend too. thought there could be a way to do it without any firmware changes. I guess thanks for confirming it. sorry that there's no way to upgrade it without changing to a different firmware. for some reason, I kinda like the simplicity of the creality firmware and the touch screen. wish we could keep it if we end up changing to klipper or any other firmware.

lbreggi, 2022-11-30:
Hi All! If we had to ask Creality about the top 5 things that we want to get enabled on the official firmware, what would be?

jklap, 2022-12-07:
> In configuration.h (CRTL-F and find those line and update values)
> 
> `#define HEATER_0_MAXTEMP 300 `
> 
> `#define BED_MAXTEMP 115`

@dremc does this account for this previous comment?

https://github.com/CrealityOfficial/Ender-3S1/issues/4#issuecomment-1135334943

vlad-the-compiler, 2022-12-24:
FIY - The _touch screen module_ is closed-source, **not** the host firmware on the S1 Pro - that is still vanilla Marlin. The fancy display module is just a controller that sends/reads gcode.

That being said, there is no reason for the S1 Pro firmware to not be published.

@CrealityTech can we please get the Marlin repo for the S1 Pro already, so we can compile our own firmware modifications just like for every other model you sell?

Pethical, 2023-01-01:
Not providing the source code, what is based on Marlin, is a clear violation of the GPL and might be enforced by law. This is not a thing what can decided by @CrealityTech .

aeyjpn, 2023-01-03:
I just buy a new Ender 3 S1 Pro and found that cannot install any custom firmware with this touch screen. 
Please help us @CrealityTech.
My Ender 3 v2 is being upgrade to this direct drive extruder too!

grrmisfit, 2023-01-09:
just tell friends, post on facebook, reviews on amazon to avoid s1 pro/plus till touchscreen firmware is released

RWBRITT, 2023-01-09:
Avoiding those models won't work. It would be better to avoid all of their touch screens. I had it somewhat working. I’ll post what I have when I get a chance as far as the edits needed to make creality screen to work.

vlad-the-compiler, 2023-01-10:
@RWBRITT you've reversed the S1 Pro touchscreen protocol? Intriguing. Really curious to see your insights.

RWBRITT, 2023-01-10:
> @RWBRITT you've reversed the S1 Pro touchscreen protocol? Intriguing. Really curious to see your insights.

This is the part of marlin adv config that stands out. The adv config example doesn't show #define dgus-serial-port line. I got this from marlinfw.org. 

// dgus screen settings adv config file 
#if HAS_DGUS_LCD
  #define DGUS_SERIAL_PORT 3
  #define DGUS_BAUDRATE 115200

  #define DGUS_RX_BUFFER_SIZE 128
  #define DGUS_TX_BUFFER_SIZE 48
  //#define DGUS_SERIAL_STATS_RX_BUFFER_OVERRUNS
  #define DGUS_UPDATE_INTERVAL_MS  500
  #if EITHER(DGUS_LCD_UI_FYSETC, DGUS_LCD_UI_HIPRECY)
    #define DGUS_PRINT_FILENAME
    #define DGUS_PREHEAT_UI
    #if ENABLED(DGUS_LCD_UI_FYSETC)
      //#define DGUS_UI_MOVE_DIS_OPTION
    #else
      #define DGUS_UI_MOVE_DIS_OPTION
    #endif
    #define DGUS_FILAMENT_LOADUNLOAD
    #if ENABLED(DGUS_FILAMENT_LOADUNLOAD)
      #define DGUS_FILAMENT_PURGE_LENGTH 10
      #define DGUS_FILAMENT_LOAD_LENGTH_PER_TIME 0.5
    #endif
    #define DGUS_UI_WAITING
    #if ENABLED(DGUS_UI_WAITING)
      #define DGUS_UI_WAITING_STATUS 10
      #define DGUS_UI_WAITING_STATUS_PERIOD 8
    #endif
  #endif

vlad-the-compiler, 2023-01-10:
If fiddling with the pre-existing DGUS (DWIN)-based external UIs would get the S1 Plus / Pro touchscreen to work, that would be quite interesting. It would not only mean that they've essentially just based it on what appears to be some sort of standard, but it would also enable people to roll custom Marlin compilations without requiring a display replacement.

The actual usefulness of the touch screens isn't that high up sadly, as they offer a very limited subset of features compared to the E3 V2 / S1 vanilla knob display, classic 128x64 panels or even plain gcode over serial.

RWBRITT, 2023-01-10:
> If fiddling with the pre-existing DGUS (DWIN)-based external UIs would get the S1 Plus / Pro touchscreen to work, that would be quite interesting. It would not only mean that they've essentially [...]

Using the example config files for a particular printer, I've found some are not even close to being correct. Like the example that is out there for the ender 3 S1 or V2's, don't have anything about assigning a serial port to the display. I thought about it and dug into MarlinFW site and found the problem with the adv config missing the #define dgus_serial_port. It's just 2 computers needing to communicate. Should be easy peas but "NOPE!"  🤣

Pethical, 2023-01-10:
The touch screen is good innovation, but not in this way, not this cheap HMI displays.
This is why I don't understand Creality why don't want to give out the code. Basically it's simple. Designed UI, simple commands with 115200 baud. Maybe not worth to check this protocol, a new hmi project and modify Marlin to handle it is faster. The DWIN designer is free, anybody can make a new UI for it. But the best thing is pull out, connect an other marlin compatible display, then use it ín your iot projects. No hmi display in your printer, no problems, no more Creality and you can use a legal firmware.

vlad-the-compiler, 2023-01-11:
@Pethical I'd go one step further and say that since full control is possible via just gcode over serial (down to the point of even emulating 'Wait for user functionality' with `M108`), there is no reason to have displays that implement their own UI with custom protocols.

A few months back I've dabbled with the idea of a universal display for Marlin that talks to the printer over UART and provides a graphical shell and Wi-Fi capability through ESP3D (similar to Bigtreetech's offering, yes, but ideally more flexible), and I wrote it off thinking that surely 3D printer manufacturers are already doing a way better job at this point - but seeing how degraded in functionality these "modern" UIs have got, I'm starting to seriously consider that project.

Pethical, 2023-01-12:
Yes, you are true. That would be an excellent thing.  Convenient, user-friendly, but not as limiting as smart (dumb) hmi. 

This github is pretty dead, recently it was said that Creality is an Open Source evangelist. Compared to this, there is no life, no answers, closed source code, stupid solutions, processor replacements. What's up Creality? Does this company still exist at all, or have you closed it down?

RWBRITT, 2023-01-12:
I don't think @creality cares. Their support of their machines have been terrible. I've sent help requests and no answer. I've seen other people complaining about lack of responses to their requests as well, on other forums.

booperscoper, 2023-01-14:
I found this thread, and made an account just to comment.  I chatted with the Tech Support last night (CA, 12:30am).  I was told they plan on releasing after the New Year (Chinese New Year).  Don't know if any of you want to go to the Creality homepage, and do the same; see if you get the same answer.   I also asked if I could make changes to the S1 config to make it like an S1 Pro in my case, but I was told no and I didn't ask why.

Wasn't sure if this helped anyone, but I say keep asking.  Lets be real; the source files are sitting somewhere that they (Creality) only need to make a hyperlink for to share.

Happy New Year everybody!

booperscoper, 2023-01-14:
Hey everybody, what's this?  Thank you anyone for your help!!

https://github.com/CrealityOfficial/Ender-3S1/pull/35

booperscoper, 2023-01-14:
@mriscoc Love your firmware sir, I wish I could just get it to work with Ender 3 S1 Pro sir.  Can't  get the leveling to work, and I went back to stock.  Same problems others have mentioned on your facebook page.  Still love the way it looks though!

grrmisfit, 2023-01-15:
> 

Use a v2 screen or s1 screen and use this mod even for stock firmware https://github.com/mriscoc/Ender3V2S1/issues/614#issuecomment-1374667776

booperscoper, 2023-01-15:
Thank you so much for that mod I'm currently in the process of getting started to print it.  That will allow me to get to the very top of the bed without having it meet. It's extreme!  I'll just change the offset in the source code on the s1 firmware. Change the temperature up to 300 and will call it a day. You have a fantastic New Year

andre68723, 2023-01-15:
> Hey everybody, what's this? Thank you anyone for your help!!
> 
> #35

new in 3DP, does this mean the missing source dode is now available opening the window for a "professional" firmware?

Pethical, 2023-01-15:
@andre68723 
Yes, it means, the way is opened. You can build and install your own firmware, you can modify, etc. If you want to build, then you need my fix from #38 because one syntax error and one unclosed comment :-)

booperscoper, 2023-01-15:
@lbreggi
1) Change the Z-offset increments from 0.05 to 0.01.
2) At the start of every print, no matter what sliceer I use after it homes it homes again and touches the nozzle to the bed drives me crazy.
3) I may be a moron but I haven't found it yet. Where's the disabled steppers button / function?  I have to wait for it to time out.
4) This one's a personal preference, but I would go from a 4x4 to a 5x5 grid at stock.
5) I'm sure there's another one, but that's the only four I can think of.

wizardkitty, 2023-01-15:
Well this issue has been resolved.

booperscoper, 2023-01-15:
My opinion of course. I mean no offense to wizard kitty I wouldn't call it resolved because it still requires s1 pro owners like myself to buy something else.  To your point though many many workarounds but I still would like to use the touch screen. I know I already said I bought screens but I shouldn't have to have or someone with less experience which I have little of having to buy. Something is a little silly to me.  My opinion of course

wizardkitty, 2023-01-15:
> My opinion of course. I mean no offense to wizard kitty I wouldn't call it resolved because it still requires s1 pro owners like myself to buy something else. To your point though many many [...]

They actually released the source code. So yes touchscreen support included

ayushsharma82, 2023-01-15:
Creality has made the source public for S1 Pro with `s1_pro` branch 😄and it looks like that it contains source of touch screen as well. I think we can create new UI within C++ so that's an advantage. I'll soon be reviewing the new branch and start adding improvements. 

@CrealityTech Thanks for the effort, I request people to build upon this and improve it rather than creating forks.

RWBRITT, 2023-01-15:
> Creality has made the source public for S1 Pro with `s1_pro` branch 😄and it looks like that it contains source of touch screen as well. I think we can create new UI within C++ so that's an [...]

the example config files are not real clear on what is the correct settings for that screen. do you happen to have it figured out?

ayushsharma82, 2023-01-15:
Nope, I just took a small look at the changes. I think this [dwin](https://github.com/CrealityOfficial/Ender-3S1/tree/s1_pro/Marlin/src/lcd/dwin) folder is related to the touch screen.

Pethical, 2023-01-15:
@RWBRITT 
```//
// Creality V4.2.5 display. Creality board but retain the display.
//
#define RET6_12864_LCD

// resistive touch screen
#define RTS_AVAILABLE
```
and the code in:
Marlin/lcd/dwin/lcd_rts*

@wizardkitty you can always define 1x1 pixel sized images, and you have a graphics display 😄 And characters as images and you have a character display.

If tried to do a 8x8 mesh on my S1 Pro with bilinear, but it works only if I switch off the BLTouch High Speed Mode, with high speed mode it works only until 40 points, at the 40. point it stops. With slow mode it creates a mesh, but when it saves to the eeprom the watchdog timer kills it. 😄

@CrealityTech Thank you very much, this is a big deal! As we can we will improve it and contribute.

@ayushsharma82 I hope Marlin will integrate it and we can upgrade to 2.1.x Marlin 😺

booperscoper, 2023-01-16:
I'm sorry for my ignorance everybody. Can someone put a link to the s1 pro source code? I don't know why I can't find it and it must be obvious and I can't figure it out. And I'm sorry. Thank you anyone for your help.

wizardkitty, 2023-01-16:
@RWBRITT that's community firmware.

@booperscoper use #38 for now. Creality gave us some broken code but thanks to the person who fixed it I can confirm it WORKS!

Pethical, 2023-01-16:
This DWIN display is really crazy. I tried to create a new UI based on the original. Brrr. The designer software is horrible, the half of the softwares are traditional chinese... I have experience with normal displays and nextion hmi-s, but I can't create a basic page what works on it in a day. 💩 I give it up for now, this is not my favourite display. 😄 Easyer to make an UI with a normal not "smart" touch lcd. I have a lot of sympathy for the developers who work with this, and I'm sorry that they have to go through this every day on the job. It can be terribly frustrating. 😃

jklap, 2023-01-16:
> This DWIN/DACAI display is really crazy. I tried to create a new UI based on the original. Brrr.

@Pethical Any chance you want to document what you’ve done thus far somewhere?

Pethical, 2023-01-16:
Ok, we are friends with the designer software. 😄 Here is the initial project: https://github.com/Pethical/Ender-3-S1-Pro-Screen

**At this time only the splash screen working and some page transations, so it's not usable jet, do not try it. IT IS NOT WORKING YET!!!!**

There is a lot of touch control and settings, so it's time to do. If sameone want to help do not histate to PR.

booperscoper, 2023-01-16:
I've been seeing a lot of videos on chatGPT, what happens if you put the code into chat GPT see if it's spits you anything. Just a thought.  I'm not too confident chat. GPT knows that form of code but it may.  Thank you. All who helped above it helped me a lot. I'm going to start goofing around next weekend. I got a full week of work ahead

booperscoper, 2023-01-16:
If anything chat GPT should be able to translate the Chinese to English for you. I think very easily though

Pethical, 2023-01-16:
@booperscoper 
ChatGPT is good thing, but if you don't know what you are doing, if you don't understand the code and the effect of your modifications, then do not touch it. It is dangerous, you can crash your printer, cause fire, or brick your device. This firmware is works for a lot of users. Be carefull.

Yes, it can translate Chinese to english as a lot of tool. But when I would like to do anything in my free time (I have a small amount only) then I like to be productive. I installed 4 version of the tools. The 5.10 is the more stable from the amp website. The others are crashing and half chinese. But the 5.1 has no emulator, so I need to use a 8.1, save the project in the 5.1 then load in 8.1, then start the emulator, but the emulator needs virtual com port, so I installed a virtual serial port. Like a success, but the 5.1 has a bug, and sametimes not saves the project. This is why I can only to make an simple basic project in one day and I tired because this.

booperscoper, 2023-02-02:
Hello everybody and thank you for your help and pointed me to the direction to the ender 3S1 pro source code.  The source code totally works, if I load in the VS studio and just hit build.  I'm running into an issue of my own creation; I know for certain I don't have an F1 chip. I have an F4 chip when I go to change the development environment in the platform.ini file, and hit build. I create errors. Does anyone have any suggestions on what I'm doing wrong? I have successfully created other firmwares. I just don't know what I'm currently doing wrong. Thank you for everyone for your help.

kamoteshake, 2023-02-03:
@booperscoper I think it's better to open a new issue. please, share the error that you're getting.

kamoteshake, 2023-02-07:
@joepadmiraal #41 is the latest PR that has all of the fix. maybe try that?

booperscoper, 2023-02-11:
@joepadmiraal Did you have any luck?  I was running into issues as well but there's something very different than I needed to figure out that again the error I was causing.  As someone stated above we should probably start a new forum and I would not hijack this one, so if you need any help hit me back and we'll start a new post but I was running into issues all my own doing and it works 100%.  Again, thank you everyone for your patience and help to me. I appreciate it.

joepadmiraal, 2023-02-11:
@kamoteshake I had to do a [very small change](https://github.com/Pethical/Ender-3S1/pull/5) in the code, on top of Pethical's fixes. That version is working fine now on my ender 3 s1 plus, in combination with the standard display.

JohnScottUK, 2023-04-11:
I notice that https://github.com/CrealityOfficial/Ender-3S1/pull/35 is for V2.0.8.24, but https://www.creality.com/pages/download-ender-3-s1-pro?spm=..page_1934481.products_display_1.1&spm_prev=..page_1934481.header_1.1 indicated. V2.0.8.26 is available. Can V2.0.8.26 be published?

wizardkitty, 2023-04-12:
> I notice that #35 is for V2.0.8.24, but https://www.creality.com/pages/download-ender-3-s1-pro?spm=..page_1934481.products_display_1.1&spm_prev=..page_1934481.header_1.1 indicates V2.0.8.26 is available. Can V2.0.8.26 be published?

I guess I might need to look at doing several things if Creality doesn't still understand the GPL Lisence.  

Also any differences between .26 and .24?

kamoteshake, 2023-04-17:
> I guess I might need to look at doing several things if Creality doesn't still understand the GPL Lisence.
> 
> Also any differences between .26 and .24?

yeah. I've emailed them and contacted them about releasing the .26 but never got any response. I even asked for the Plus since Plus got the .26 first before the Pro IIRC. 

I don't think there are any. I read that the difference was for detecting the laser module automatically. I haven't verified though.