CrealityOfficial/Ender-3S1 #28: Resume print not working

lin-ycv, 2022-11-29:
Had a filament run out over night, in the normal fed new filament, pressed resume, it purged, but didn't resume printing.

kamoteshake, 2022-11-29:
which firmware and printer were you using?

lin-ycv, 2022-11-30:
2.0.8.15F4
[image]

kamoteshake, 2022-12-01:
Hello. sorry it took a while. I just tested this. Started a print then I cut the filament. After the printer detected the filament "ran out", I loaded new filament. It waited to go up the temp and then it started to prime the nozzle. It then continued printing. 

I have an Ender 3S1 Plus and my firmware is 2.0.8.23F4. I see that the latest firmware is 2.0.8.24F. maybe try updating it?

lin-ycv, 2022-12-01:
I'm not sure what is issue was, I just tested it, and it also works fine.
does the matter how long it was idle for? or maybe i miss click something...

Which website should I get the update from? There's a bunch of "Creality" websites offering firmware, are they all the same?

kamoteshake, 2022-12-01:
hahaha yeah. trust me it took me a while to find a good one. I had to contact customer support and they pointed me to the [creality cloud](https://www.crealitycloud.com/software-firmware/firmware?type=2)

closed 2023-01-10 by lin-ycv