CrealityOfficial/Ender-3S1 #29: No support for stock touch screen on Ender 3 S1 Plus.

RWBRITT, 2022-12-07:
I have an s1 with factory installed sprite and was wanting to set max temp to 300°. That's the easy part. Hard part is trying to find the right way to configure the firmware to use the stock touch screen.

wizardkitty, 2022-12-12:
Saddly there is currently no way todo this without asking creality todo it for you. Creality is currently keeping their screen closed source.

RWBRITT, 2022-12-25:
It's not really impossible. The screen is not really theirs, it's Amp's. I just want the correct config file settings they used to attach the screen to begin with.

rocket59, 2022-12-31:
+1 to this. As it's Marlin under the hood why isn't the board firmware available under the terms of the open source licence? No one needs to build the screen firmware to increase the hot end max temp.

Pethical, 2023-01-01:
Just throw out to the trash the Creality touchscreen and the mainboard (this is the correct place for products what violates GPL). Purchase a new touchscreen and mainboard from a brand who respects the licenses and laws, then install Marlin. You are done and never look back.

rocket59, 2023-01-01:
Hopefully the Marlin devs will firmly remind them of their licensing obligations (I'm sure Creality would not like to create new board firmware solely for their touchscreen printers.)

The other alternative is of course to either buy their non touch display and use this in place of the touch screen and install a Marlin variant or ditch the screen entirely and just go klipper (which has the added advantage of enabling pressure advance on stock boards).

Pethical, 2023-01-01:
This infringes on the rights of not only Marlin developers. The GPL states that you, as a user, have the right to get the full functional source, modify it, and do whatever you want with it. If you don't get it (as in this case), then your rights are also being violated. @Creality-mandy sayed earlyer that this S1 FW is similar than the Pro and Plus, but without modifications what needs for the touchscreen. This is a key, because this is a Marlin based GPL licensed FW, so the Pro and the Plus FW must be GPL licensed.
Furthermore, if the source code of the screen is integrated with Marlin, i.e. it is statically linked to it, or does not work without it, or contains even just a single line of code from it, then there is no choice, because according to the GPL it also inherits the license, so must be published.

A lot of company don't understand what is GPL and the users and costumers must known and enforce their rights againts this companies. We payed for something, and we need this thing as is, without violations of our rights. Today we have a lot of choice, and must be the right choice when we purchasing a product. I will never purchase or recommend a product what violates my rights over violation of a license and this was the last time when I purchased any product from Creality.

_PS: I don't understand why is a big thing a touchscreen. You can make a similar touchscreen in 4 hours with a Nextion smart display. Not a rocket science, I don't understand why it is a top secret at Creality. Maybe the developers sold it as a big thing to the management?_

RWBRITT, 2023-01-10:
It has to do with this part here. There are resources for menu editing on the original screen manufacturer's site. It requires windows OS.

// dgus screen settings adv config file 
#if HAS_DGUS_LCD
  #define DGUS_SERIAL_PORT 3
  #define DGUS_BAUDRATE 115200

  #define DGUS_RX_BUFFER_SIZE 128
  #define DGUS_TX_BUFFER_SIZE 48
  //#define DGUS_SERIAL_STATS_RX_BUFFER_OVERRUNS
  #define DGUS_UPDATE_INTERVAL_MS  500
  #if EITHER(DGUS_LCD_UI_FYSETC, DGUS_LCD_UI_HIPRECY)
    #define DGUS_PRINT_FILENAME
    #define DGUS_PREHEAT_UI
    #if ENABLED(DGUS_LCD_UI_FYSETC)
      //#define DGUS_UI_MOVE_DIS_OPTION
    #else
      #define DGUS_UI_MOVE_DIS_OPTION
    #endif
    #define DGUS_FILAMENT_LOADUNLOAD
    #if ENABLED(DGUS_FILAMENT_LOADUNLOAD)
      #define DGUS_FILAMENT_PURGE_LENGTH 10
      #define DGUS_FILAMENT_LOAD_LENGTH_PER_TIME 0.5
    #endif
    #define DGUS_UI_WAITING
    #if ENABLED(DGUS_UI_WAITING)
      #define DGUS_UI_WAITING_STATUS 10
      #define DGUS_UI_WAITING_STATUS_PERIOD 8
    #endif
  #endif

rocket59, 2023-01-10:
So if we include this snippet in the S1 config.adv.h it should compile and work ok with the screen? 

Also just curious why this requires windows though.

RWBRITT, 2023-01-10:
Windows is required for the menu editing software from the dwin/Amp website. Amp is the actual company that made these displays for various uses. That sample code is just part of what is needed to get a screen to connect to main pcb with marlin on it.

rocket59, 2023-01-10:
Thanks. I thought it looked a bit sparse :).

RWBRITT, 2023-01-10:
> Thanks. I thought it looked a bit sparse :). 

This link has more info and resources pertaining to our problem. 

https://github.com/CR6Community/cr6community.github.io

wizardkitty, 2023-01-15:
Creality just released firmware and #38 can be compiled.

rocket59, 2023-01-15:
So the f/w for the S1 Plus is identical to the S1 Pro (other than the config settings?)

wizardkitty, 2023-01-15:
@rocket59 Yes. Infact it includes the plus. Just gotta uncomment one thing and comment another. But basically yes.

rocket59, 2023-01-15:
Cool thanks @wizardkitty and @CrealityTech for releasing the code.

jmachuca77, 2023-02-08:
@wizardkitty I tried flashing the firmware I successfully built, but the screen stays with the start of the splash screen and never shows anything. Can you elaborate on how the code in the s1_pro branch can be used? My printer came with SWV2.0.8.25F4 from the factory and the available software on the branch is labeled 2.0.8.24F4 I am guessing its a firmware mismatch why my code doesn't show anything on the display?

jmachuca77, 2023-02-09:
@wizardkitty Never mind I got it working! I missed changing the processor define to the 401, once I changed that I is now working.