CrealityOfficial/Ender-3S1 #31: Reminder of the terms of the GNU General Public License v3.0

rocket59, 2023-01-01:
"_Permissions of this strong copyleft license are conditioned on making available complete source code of licensed works and modifications, which include larger works using a licensed work, under the same license. Copyright and license notices must be preserved. Contributors provide an express grant of patent rights_."

from https://github.com/MarlinFirmware/MarlinDocumentation/blob/master/LICENSE

wizardkitty, 2023-01-04:
Creality seems to NOT understand this. I even asked support for it and they wouldn't budge even when told about the license.

There are other printers that need their source code open sourced from creality. But unless creality starts taking care of this fast there may need to be some enforcement.

Pethical, 2023-01-04:
Marlin's developers are constantly fighting these companies and occasionally succeeding, but they also need a lot of help from the community. Various influencers, blogs, and thematic sites should not recommend a product that violates these licenses. There was a similar case in 2018, when the entire community united against Creality and there were also some of the distributors that removed these products. Then Creality published the source and got another chance from the community. According to these, it did not learn from it, even though the community does not forget.

rocket59, 2023-01-15:
Thanks @Pethical. Also kudos to Creality for listening and releasing the source code for the remaining S1 printers!

closed 2023-01-15 by rocket59