CrealityOfficial/Ender-3S1 #44: Will this work with H/W VER CR-FDM-v2.5. S1_1 ?

trapper-, 2023-01-27:
https://github.com/CrealityOfficial/Ender-3S1/blob/7fffa9270ff8fb5ee208a5b04f832e20bab19c60/Marlin/Configuration.h#L108

Will this firmware work for me with this hardware version mismatch?

kamoteshake, 2023-01-27:
The one that you linked is just for the about page in the screen. for the board compatibility, you have to look at
https://github.com/CrealityOfficial/Ender-3S1/blob/7fffa9270ff8fb5ee208a5b04f832e20bab19c60/Marlin/Configuration.h#L78-L79
you might have to change these in the `platformio.ini` if you change the lines above to compile the firmware.
https://github.com/CrealityOfficial/Ender-3S1/blob/7fffa9270ff8fb5ee208a5b04f832e20bab19c60/platformio.ini#L18-L22

trapper-, 2023-01-27:
Yeah ok I see that string is for display.

Just seems odd to me if it ends up displaying a hardware version which doesn't match my actual hardware. Wonder what is the point of displaying it in the first place if it's not going to be accurate.

kamoteshake, 2023-01-27:
you have to understand that this firmware is being used by multiple machines. I have a feeling that Creality just selected a version for the S1 Pro and pushed it here. Just FYI, I was able to fork this firmware and use it on my S1 Plus. Sure I don't get the latest version of the S1 Plus firmware (since this is based one 2.0.8.24 and the latest for S1 Plus is 2.0.8.26) but I was able to configure it to be used on my S1 Plus.