CrealityOfficial/Ender-3S1 #46: PC0 pin as PWM?

kamoteshake, 2023-02-03:
Hello. Just a quick intro, I'm new to the firmware game so please treat me like a total noob. 😅

### Context
I upgraded my coldend fan to a 4020 and I'm looking for a way to slow it down. I know Klipper can slow the fan down and you can control it like it's PWM but it seems like you can't do it in Marlin. I already tried changing the `EXTRUDER_AUTO_FAN_SPEED` in the `Configuration_adv.h` to `128` (~50%) but it's always running at 100% speed.

### Things I've tried
So I've read a couple of issue about PWM in Marlin and tried to apply things but no luck.

1. Change `EXTRUDER_AUTO_FAN_SPEED` to different values. but it seems like it's always running at `255`. I've followed this chain and it brought me to this [lines of code](https://github.com/CrealityOfficial/Ender-3S1/blob/s1_pro/Marlin/src/module/temperature.cpp#L850-L855) which brought me to the assumption of `PC0` not being a PWM.
2. Following the chain of code above, it brought me to the [PeripheralPins.c](https://github.com/CrealityOfficial/Ender-3S1/blob/s1_pro/buildroot/share/PlatformIO/variants/MARLIN_CREALITY_STM32F401RC/PeripheralPins.c) of `MARLIN_CREALITY_STM32F401RC`. So, I added `{PC_0,  TIM3,   STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF2_TIM3, 1, 0)},` in the [PinMap](https://github.com/CrealityOfficial/Ender-3S1/blob/s1_pro/buildroot/share/PlatformIO/variants/MARLIN_CREALITY_STM32F401RC/PeripheralPins.c#L90-L132) for the PWM. This caused the fan to not turn on at all unless I set `EXTRUDER_AUTO_FAN_SPEED` to `255`.
3. so I commented out the `FAN_SOFT_PWM` in the [pins_CREALITY_S1_F401RC.h](https://github.com/CrealityOfficial/Ender-3S1/blob/s1_pro/Marlin/src/pins/stm32f4/pins_CREALITY_S1_F401RC.h#L163) and uncommented the `FAST_PWM_FAN` in the [Configuration.h](https://github.com/CrealityOfficial/Ender-3S1/blob/s1_pro/Marlin/Configuration.h#L2938). This didn't do anything. 

### My Questions
1. Is `PC0` a PWM and I'm just setting this wrong?
2. If it is NOT a PWM pin, how can I make it into a PWM?

### Rant
I've been pulling my hair on this for a while. I think I'm close but I also feel like I'm doing way too much. I don't know if I'm using the wrong timing or channel for the pin. It's just annoying how Klipper does it well yet Marlin doesn't. I really want to use the stock firmware as much as I could before jumping to a different platform or is that the wrong mindset? I don't know. Maybe people want to upgrade their coldend fan and would want to be able to adjust the speed of it.

Thank you!

kamoteshake, 2023-02-05:
### Update 2
So I tried installing the Marlin 2.1.2, still have the same issue with the Soft PWM. I found someone who already rebased the [S1 firmware](https://github.com/synman/Marlin/tree/2.1.2-ender-3-s1) to use Marlin 2.1.2 and the bugfix-2.1.x. I will try the bugfix-2.1.x firmware and see if the issue is still there.

I've been reading a lot of issue posts in Marlin from other board having the same behavour. I really don't know who should own this issue. When I went to Marlin support, they pretty much said that they don't have Creality's pins in the stock Marlin and left it there. Does this mean that Creality have to send something to Marlin to have the board pins in the stock Marlin? I really don't know how all of these work.

Anyways, I'll try out the bugfix branch and see.

mriscoc, 2023-02-06:
I think someone else is also working on something similar: https://github.com/mriscoc/Ender3V2S1/discussions/702

kamoteshake, 2023-02-06:
Oh. thank you for the link!

I-wish-2, 2023-02-06:
Greetings,
After a note from mriscoc I thought this information might helpful.  I was working to install a Solvo laser on my Ender 3 Pro that is using mriscoc's current firmware but I needed the parts cooling fan circuit ( PA0 ) to supply the control drive for the laser.  This circuit is a PWM one as that is the way it controls the parts cooling fan and the mother board fan.  My only issue is that in fan mode the PWM is only 7.87 Hz and supplies 0 to 24vdc.  I needed a PWM of 1KHz at 0 to 5 vdc.

My first step was to acquire a 0 to 5 vdc PWM point on the 4.2.2 or 4.2.7 board.  See the picture.  Its works on both boards.  The 100 ohm resistor is a REQUIREMENT or the MosFet gate, where the resistor is soldered will burn out.  It is in series with the wiring and is the + side of the PWM signal.  The negative side of the PWM signal can come from any 24vdc (-) connection on the board.
My second step was to change the firmware to acquire the 1KHz frequency for the PWM signal.  Working with the Marlin github link I was able to acquire the 1KHz with the edits listed below.

The PWM firmware edits:

Marlin  >  configuration.h:  Your line numbers could be different !  The lines are shown with the edits.
Comment out Line 3247  // #define FAN_SOFT_PWM   ( add the // marks to the front of the line )
*****
Marlin  >  configuration_adv.h:  Your line numbers could be different !  The lines are shown with the edits.
Edit Line 585  #define FAN_MIN_PWM 0   ( change the value here to 0 )
Uncomment Line 613  #define FAST_PWM_FAN   ( remove the // marks )
*****
Marlin  >  src  >  pins  >  stm32f1  >  pins_CREALITY_V4.h
Comment out Line 154  //#define FAN_SOFT_PWM_REQUIRED  ( add the // marks to the front of the line )
*****

" Above the Marlin Folder in Visual Studio Code "

ini  >  stn32f1.ini:   Your line numbers could be different !  The lines are shown with the edits.

After         Line 131      -DSS_TIMER=4 -DTIMER_SERVO=TIM5
Add new   Line 132     -DSS_TIMER=4 -DTIMER_SERVO=TIM8

Please note that with the PWM frequency set to 1KHz both of the cooling fans will not operate as the frequency is to high to allow them to start.  I wired my mother board cooling fan to the Hot End cooling fan green connector and it runs all the time.  The parts cooling fan is unplugged and not used as the printer is now a laser based unit not a filament printer.

I might add that keeping a flashable bin file of the original printer setup is a good idea.  And a separate fin file for what your project is.
I hope this might help.
Doug
 
[image]

I-wish-2, 2023-02-06:
Greetings,
I'm sorry that the information is not much help for you.  The reason for the PWM frequency change was that the laser was pulsing to much during a burn because the initial fan frequency of 7.87 Hz was way to low for the laser.  LightBurn suggested to move it up to about 1`KHz. 
The first edit produce an increase to 50 Hz and those at Marlin said that was because of a timer conflict with one of the axis drivers.  They suggested this additional edit regarding the timer.

ini > stn32f1.ini: Your line numbers could be different ! The lines are shown with the edits.

After Line 131 -DSS_TIMER=4 -DTIMER_SERVO=TIM5
Add new Line 132 -DSS_TIMER=4 -DTIMER_SERVO=TIM8

This resulted in the PWM frequency reaching 1KHZ which has the laser burning correctly.  I have tried working in the configuration_adv.h files at line below but was not ever successful.  I think I don't know how to edit the lines.

To Increase PWM frequency from 1KHz.
Configuration_adv.h

615 	You can optionally uncomment and modify FAST_PWM_FAN_FREQUENCY
	in Configuration_adv.h if you need value different than 1kHz.

Origional Lines 613 to 624  FAST_PWM_FAN_FREQUENCY
Lines
613	#define FAST_PWM_FAN    // Increase the fan PWM frequency. Removes the PWM noise but increases heating in the FET/Arduino
614	#if ENABLED(FAST_PWM_FAN)
615	//#define FAST_PWM_FAN_FREQUENCY 31400  // Define here to override the defaults below
616	//#define USE_OCR2A_AS_TOP
617	#ifndef FAST_PWM_FAN_FREQUENCY
618	#ifdef __AVR__
619	#define FAST_PWM_FAN_FREQUENCY ((F_CPU) / (2 * 255 * 1))
620	#else
621	#define FAST_PWM_FAN_FREQUENCY 1000U
622	#endif
623	#endif
624	#endif

A build always failed.  If you have any thoughts on these lines please let me know as I would like to use these to line and return all the others to their original settings.  This keeps the firmware simple and as I'm a novice at this type of work it would help me a lot.
Best Regards
Doug

kamoteshake, 2023-02-06:
I mean if the .ini changes worked, I think you should keep it. it's weird though that those are defined in the environment and not in the code. I guess it's because the env settings take precedence? I'm sorry, I couldn't try your changes since I don't have a laser module and my board is STM32F401RC.

kamoteshake, 2023-02-08:
I ran out of ideas. 😔 If you guys have any other ideas, let me know. for now, I'll put this on the side burner.