CrealityOfficial/Ender-3S1 #47: Adjusting Z-Offset Factor on Stock Firmware

booperscoper, 2023-02-11:
Thank you everyone who helped me successfully find and install the source code firmware for my Ender 3 S1 Pro. 

My question is--how do I change the increments in which the Z-Offset Baby Step moves by?  For example, it currently changes .05mm a step.  Does anyone know how to change it to .01mm?  I have changed it before in other firmware, and it always works.  Not in the Ender 3 S1 Pro Fixes Firmware. 

 I've  tried:

#define BABYSTEPPING
#if ENABLED(BABYSTEPPING)
  //#define INTEGRATED_BABYSTEPPING         // EXPERIMENTAL integration of babystepping into the Stepper ISR
  //#define BABYSTEP_WITHOUT_HOMING
  #define BABYSTEP_ALWAYS_AVAILABLE       // Allow babystepping at all times (not just during movement).
  //#define BABYSTEP_XY                     // Also enable X/Y Babystepping. Not supported on DELTA!
  #define BABYSTEP_INVERT_Z false           // Change if Z babysteps should go the other way
  #define BABYSTEP_MILLIMETER_UNITS       // Specify BABYSTEP_MULTIPLICATOR_(XY|Z) in mm instead of micro-steps
  #define BABYSTEP_MULTIPLICATOR_Z  0.01       // (steps or mm) Steps or millimeter distance for each Z babystep
 // #define BABYSTEP_MULTIPLICATOR_XY 1 

and I've also tried:

#define BABYSTEPPING
#if ENABLED(BABYSTEPPING)
  //#define INTEGRATED_BABYSTEPPING         // EXPERIMENTAL integration of babystepping into the Stepper ISR
  //#define BABYSTEP_WITHOUT_HOMING
  #define BABYSTEP_ALWAYS_AVAILABLE       // Allow babystepping at all times (not just during movement).
  //#define BABYSTEP_XY                     // Also enable X/Y Babystepping. Not supported on DELTA!
  #define BABYSTEP_INVERT_Z false           // Change if Z babysteps should go the other way
  //#define BABYSTEP_MILLIMETER_UNITS       // Specify BABYSTEP_MULTIPLICATOR_(XY|Z) in mm instead of micro-steps
  #define BABYSTEP_MULTIPLICATOR_Z  8       // (steps or mm) Steps or millimeter distance for each Z babystep **(Stock is 40)**
 // #define BABYSTEP_MULTIPLICATOR_XY 1   

Any help would be great.  Thank you!

kamoteshake, 2023-02-16:
@booperscoper Were you able to get this working? if not, have you checked the [lcd_rts.cpp](https://github.com/CrealityOfficial/Ender-3S1/blob/s1_pro/Marlin/src/lcd/dwin/lcd_rts.cpp#L1612-L1639)? Check the `zprobe_zoffset`

wizardkitty, 2023-07-02:
Saddly this is part of the control screen. They luckily made that open source.