CrealityOfficial/Ender-3S1 #50: Open source announcement

ThomasToka, 2023-06-14:
Dear Creality,

after your announcement in social media regarding "open source" we are missing so many not released source code!

Where is it? Why is it not automatically pushed to github after a new release? 

Your PR was good. But we are still waiting.

Please do what you announced: OPEN SOURCE your code please. Full and complete.

Thank you.

Thomas

wizardkitty, 2023-07-02:
This is infact REQUIRED by Creality. It required demands to get the S1 Pro and Plus Firmware. All firmware MUST be made opensource if its marlin. and its pretty well known it is marlin.

ThomasToka, 2023-07-02:
hej @wizardkitty i forked, ported and extended it already.. https://github.com/ThomasToka/MarlinFirmware/releases/tag/MARLIN-E3S1PROFORK-BYTT-v001

maybe you like it :)