CrealityOfficial/Ender-3S1 #51: A way to downgrade screen firmware (Ender 3 S1)?

Crapkin22, 2023-06-23:
After checking out  the last official firmware update for the Ender 3 S1 (Marlin2.0.8_Ender-3S1_HWv24S1_301_SWV3.1.9_Two_C_F401_FDM_LASER), I wanted to go back to "mriscoc" firmware. It seems the latest screen firmware update (one  that includes the "dcboot" file) has made any other screen firmware incompatible. And the screen just stays black.
I tried downgrading to older versions by just flashing older updates, but it didnt seem to help. So I was wondering if there is a way to downgrade to stock firmware for the screen. 
Thanks in advance.

kamoteshake, 2023-07-02:
you should be able to just download older versions and just flashing the screen firmware.

Crapkin22, 2023-07-02:
Doesnt work unfortunately. I was able to just flash old screen firmware and
make that work with mriscoc. But the screen breaks if you upload anything
other than official creality firmware.

kamoteshake, 2023-07-03:
S1 screen has the knob. right? I thought that was supported by mriscoc? 🤔

HK8088, 2024-06-14:
im having the same issue