CrealityOfficial/Ender-3S1 #52: Continued Violations of Open Source Agreements

wizardkitty, 2023-07-02:
Creality. You appear to still be pushing out new firmware without uploading the source code. This needs to stop.

You are violating the GNU General Public License v3.0 that you agreed to by not disclosing the source code.

To anyone else: Please report Creality by following the instructions here. https://www.gnu.org/licenses/gpl-violation.en.html

wizardkitty, 2023-07-07:
Aparently Creality thinks that the source code used to be confidential when it wasn't the moment the printer was released.

Claiming that the marlin source code used to be confidential over a year after printer release isn't a good excuse for delaying giving source code.

wizardkitty, 2023-07-07:
Alright. I can confirm creality is in violation of parts of the lisence. They do not give out the gpl-3 lisence as they are required to with any download of the firmware.

kamoteshake, 2023-07-11:
> Claiming that the marlin source code used to be confidential over a year after printer release

This is confusing. So they think that the code after a year of release becomes confidential? 🤔 

Even if that is true, then with that logic, they already broke their logic. Ender S1 Plus was released last year and there were a bunch of updates with in the first year and they didn't release the source code of those updates. 🤷🏼‍♂️ Either way. they're pretty much violating licenses.

wizardkitty, 2023-07-11:
> This is confusing. So they think that the code after a year of release becomes confidential? 🤔
>
> Even if that is true, then with that logic, they already broke their logic. Ender S1 Plus was released [...]

Well they are still claiming it was confidential after this long of a time and it seems like a weird excuse.

wizardkitty, 2023-08-15:
@CrealityTech This must be fixed.