CrealityOfficial/Ender-3S1 #53: Ender S1 Pro - no leading zeroes on minutes in print progress monitor

sadgit, 2023-07-04:
version 2.0.8 26F4

When displaying progress "One Hour and Five Minutes" is displayed as 1:5

It should be displayed as 1:05

Thanks

wizardkitty, 2023-07-08:
With how the S1 Pro firmware is this would have to be in the screen firmware.

ThomasToka, 2023-07-12:
The bug here indeed is in the screen firmware here on sites 1,9,10 and 12. "Display invalid zeros" is deactivated for this field.

I have already fixed this in my development branch an will deliver this fix with my v002 release.

[image]

ThomasToka, 2023-07-12:
this is how it looks on this site in the editor:

[image]

jklap, 2023-07-12:
I don't see any screen files in your repo? Where are you keeping these types of changes @ThomasToka ?

ThomasToka, 2023-07-12:
v002 is not released yet. you can download v001 with its screen with many enhancements in my v001 release https://github.com/ThomasToka/MarlinFirmware/releases/tag/MARLIN-E3S1PROFORK-BYTT-v001

ThomasToka, 2023-07-12:
yes sure. naturally i release all source files after my final release.. https://github.com/ThomasToka/Marlin/tree/MARLIN-E3S1PROFORK-BYTT/ScreenFirmware

if you want to compile the v001 state you need to take this https://github.com/ThomasToka/Marlin/commits/MARLIN-E3S1PROFORK-BYTT-RELEASE-CODE . other branches are mid development as i continue developing  ;)

ThomasToka, 2023-07-15:
today i released v002 with the mentioned fix, ubl meshviewer and other fixes. I created a new branch for the release @wizardkitty.

Naturally all in the release. Including the screen sources.