CrealityOfficial/Ender-3S1 #55: Warnig src_filter

Fredlepp, 2023-09-25:
Hello, what should I do to solve this problem?

Warning! Ignore unknown configuration option `monitor_flags` in section [env]
Warning! `src_filter` configuration option in section [env:DUE] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_filter` configuration option in section [env:esp32] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_build_flags` configuration option in section [env:linux_native] is deprecated and will be removed in the next release! Please use `build_src_flags` instead
Warning! `src_filter` configuration option in section [env:linux_native] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_filter` configuration option in section [env:SAMD51_grandcentral_m4] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_filter` configuration option in section [env:malyan_M300] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_filter` configuration option in section [env:teensy31] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_filter` configuration option in section [env:teensy35] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_filter` configuration option in section [env:teensy36] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_filter` configuration option in section [env:teensy41] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! Ignore unknown configuration option `monitor_flags` in section [env]
Warning! `src_filter` configuration option in section [env:DUE] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_filter` configuration option in section [env:esp32] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_build_flags` configuration option in section [env:linux_native] is deprecated and will be removed in the next release! Please use `build_src_flags` instead
Warning! `src_filter` configuration option in section [env:linux_native] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_filter` configuration option in section [env:SAMD51_grandcentral_m4] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_filter` configuration option in section [env:malyan_M300] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_filter` configuration option in section [env:teensy31] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_filter` configuration option in section [env:teensy35] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_filter` configuration option in section [env:teensy36] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_filter` configuration option in section [env:teensy41] is deprecated and will be removed in the next release! Please use `build_src_filter` instead

Pethical, 2023-09-25:
Replace all src_filter with build_src_filter or use a previous version of PlatformIO, or ignore this warnings.

Fredlepp, 2023-10-21:
it can compil with warning src_filter, but printer or screen doesn't want to boot