CrealityOfficial/Ender-3S1 #60: Please provide the latest source code

CalDymos, 2024-01-25:
Please provide the current source code, this is the only way the firmware can be improved and extended by the community.
Let's improve and optimize the firmware together.
This is how open source works.

CalDymos, 2024-02-22:
Is anyone from Creality Support still active here?
Please upload the current source code of the firmware.

necrolingus, 2024-02-23:
I'm wondering the same thing. It was Chinese New Year so that could have an impact, but it seems in general the Creality team is not on top of their game anymore. I posted my bug on the Creality forums and after no response I posted it here on GH, but both places no response. The guys on the Cura forums are way more active and tried helping out immediately, but unfortunately it seems like it is an S1 pro bug where the latest firmware (on Creality's website) completely ignores slicer speed and accaleration settings.

CalDymos, 2024-02-23:
Support ([cs@creality.com](url)) told me that technical support is only active on Facebook ([https://www.facebook.com/crealitysupport](url)).  I'll give it a try there too. Also try a request via Facebook, I think the more requests for the current source code, the better.

Aenopix, 2025-05-19:
Any update on this? I'll also try contacting them for the updated firmware version.

FroggieFrog, 2025-05-19:
Don't wait. Just use this https://github.com/ThomasToka/MarlinFirmware and be happy.